cattaneod/CMRNet #2: Is training correct?

oishigyunyu, 2020-07-10:
Hello, Cattaneo.

I'm trying to your this project, and it's in process of training on my PC. So, I have a question.
In the training of CMRNet, the transformation error doesn't decrease such as your paper.
Is my training correct?
I tried below command referenced as this repository README.
`python main_visibility_CALIB.py with batch_size=24 data_folder=./KITTI_ODOMETRY/ epochs=300 max_r=10 max_t=2 BASE_LEARNING_RATE=0.0001 savemodel=./checkpoints/ test_sequence=0`
And putting result.
[image]

cattaneod, 2020-07-13:
Hi,
I just tested the code with the following command:
```
python main_visibility_CALIB.py with batch_size=24 data_folder=./KITTI_ODOMETRY/ epochs=300 max_r=10 max_t=2 BASE_LEARNING_RATE=0.0001 savemodel=./checkpoints/ test_sequence=0 seed=42
```
and it's working fine, the loss should reduce to ~ 1.25 after the first epoch.
Maybe it's a data problem? i see that you have 477 batches, while I have 464 batches.

[image]

oishigyunyu, 2020-09-15:
Sorry for being late for replying to cattaneo. My professor  helped me for this training problem, and I could solve this.
But I remember that I encountered the assertion error about the coordinate of the data.
**The error message was “Wrong Coordinates”(at camera_model.CameraModel.project_pytorch()).**
I think this was configured with your intent. So is this the wrong?

kuanchihHuang, 2021-03-25:
Hi, @cattaneod ! Thanks for your excellent work! I found that I also have 477 batches instead of 464 batches. Can you help me to figure out the reason? Thanks a lot.

cattaneod, 2021-03-30:
Hi @kuanchihHuang,

I just downloaded the KITTI odometry dataset again and run the preprocessing, and 477 batches is indeed correct.
The loss in the first few epochs looks very similar to the one I used for the paper, so you should get very similar results anyway.

closed 2021-12-06 by cattaneod