cattaneod/CMRNet #4: later versions of torch

Sidu28, 2021-08-16:
Will later versions of pytorch (say 1.4 and up) be compatible with the CMRNet code? Thanks

cattaneod, 2021-08-19:
@Sidu28 Thank you for your interest in our work!

The code should be compatible with newer version of Pytorch without problems. However, I can't test it right now, but I can do it next week.

Let me know if you have any problem.

egirgin, 2021-12-04:
I get the error below and the only [solution](https://stackoverflow.com/questions/68164565/cublas-runtime-error-the-gpu-program-failed-to-execute-at-pytorch-aten-src-th) I found online is to upgrade the version of the torch. However, I could not run the correlation layer after upgrading. My torch is running without problem with last stable version. Is there anyone check with the last stable version ? (current its 1.10 for torch)
[image]

cattaneod, 2021-12-06:
@egirgin I suppose that you updated PyTorch after installing the correlation package. in that case, you need to install again both `correlation_cuda `and `visibility` packages.

I just tested with PyTorch 1.10 and cuda 10.2 (cuda 11.x is not supported at the moment), and it works.
However I had to update the fille `models/CMRNet/correlation_package/setup.py` by replacing `-std=c++11` with `-std=c++14`.

Let me know if you are able to solve the problem.

egirgin, 2021-12-08:
@cattaneod I reinstalled everything from scratch on a different machine with cuda 10.2 and torch 1.10 and the error I get changed. Now I get 
```
ImportError /home/.../models/correlation_package/correlation_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN2at19UndefinedTensorImpl10_singletonE
```
Can you specify which dependencies should I replace ? 
I changed the followings 
```
cupy_cuda90==6.0.0 to cupy-cuda102
torch==1.0.1.post2 to torch
torchvision==0.2.2 to torchvision
```
Besides, I installed ```cudatoolkit=10.2``` as it suggested in the pytorch [getting started](https://pytorch.org/get-started/locally/) section.

should I change the torch version in the ```/models/correlation_package/pyproject.toml``` or ```nvcc_args``` in ```/models/correlation_package/setup.py```?? I'm not familiar with the cuda layers tbh. 

thanks in advance

egirgin, 2022-01-10:
@cattaneod I have downgraded my ubuntu version to 16 and cuda to 9.0. However, the problem persists. It is directly related with the CMR correlation module. I install it with the following command:
```bash
pip install -e ./models/CMRNet/correlation_package
``` 
It successfully installs it,
Then if I call
```python
from models.CMRNet.correlation_package.correlation import Correlation
```
in any python script, it gives the following error:
```bash
.../lib/python3.6/site-packages/correlation_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN2at19UndefinedTensorImpl10_singletonE
```

My setup is:
```
Ubuntu 16
Cuda 9.0
GTX 960M
Anaconda Python 3.6
```
Idk what is the problem... Can anyone help???

closed 2022-01-21 by cattaneod

WJ1214, 2022-05-18:
> ## **RESOLVED:EDIT**
> The solution is to import torch before correlation package
> **OLD ISSUE** @cattaneod I reinstalled everything from scratch on a different machine with cuda 10.2 and torch 1.10 and the error I get changed. Now I get
> 
> ```
> ImportError /home/.../models/correlation_package/correlation_cuda.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN2at19UndefinedTensorImpl10_singletonE
> ```
> 
> Can you specify which dependencies should I replace ? I changed the followings
> 
> ```
> cupy_cuda90==6.0.0 to cupy-cuda102
> torch==1.0.1.post2 to torch
> torchvision==0.2.2 to torchvision
> ```
> 
> Besides, I installed `cudatoolkit=10.2` as it suggested in the pytorch [getting started](https://pytorch.org/get-started/locally/) section.
> 
> should I change the torch version in the `/models/correlation_package/pyproject.toml` or `nvcc_args` in `/models/correlation_package/setup.py`?? I'm not familiar with the cuda layers tbh.
> 
> thanks in advance

Hi! I meet the same problem, have you solved it?