cattaneod/CMRNet #6: Training results cannot be reproduced

liyang159357, 2022-04-24:
In accordance with "python main_visibility_CALIB.py with batch_size=24 data_folder=./KITTI_ODOMETRY/ epochs=300 max_r=10 max_t=2 BASE_LEARNING_RATE=0.0001 savemodel=./checkpoints/ test_sequence=0"

My training results are as follows：
total test loss = 1.449
total traslation error: 164.9663721087535 cm
total rotation error: 6.641193625108441 °
------------------------------------
This is 300-th epoch

Is there something wrong with the submitted code?

liyang159357, 2022-04-25:
您好！您的来信我已经收到，我会尽快给您回复。谢谢！——李杨

liyang159357, 2022-04-26:
The program runs on Windows platform,recompile "correlation_cuda","visibility" by vs2019.
In the code，RT = T@R instead of RT = T * R. Because blender has adjusted its mathutils module, replacing the asterisk * with the at symbol @
In the config,I rename ”kitti-00.csv“ to "poses.csv" in "./KITTI/sequences/00"

caocheng1cc, 2022-11-04:
> 你可以告诉我怎么实现的吗，因为我没有找到pose.txt

caocheng1cc, 2022-11-05:
> In accordance with "python main_visibility_CALIB.py with batch_size=24 data_folder=./KITTI_ODOMETRY/ epochs=300 max_r=10 max_t=2 BASE_LEARNING_RATE=0.0001 savemodel=./checkpoints/ test_sequence=0" [...]

请问你的torch版本和torchvision版本是多少，我一直报rotat的错误

Zhaohx424, 2022-11-21:
I use the pretrained model provided by the author and evaluate it in KITTI odometry datasets, but I got a very worse result than the results in the paper. For example, the median Transl.Error tested in sequence 00 is 19.2366cm, and even in the sequence 03 which used to train the model it still 10.6865cm, they are all much bigger than the results, I want to know why?
In addition, my environment is Ubuntu 18.04, cuda11.3, PyTorch1.10, Python3.6, they are all fit.

Zhaohx424, 2022-11-21:
> I use the pretrained model provided by the author and evaluate it in KITTI odometry datasets, but I got a very worse result than the results in the paper. For example, the median Transl.Error tested [...]

OK I know, the results you put in Github uses cm, and the results in your paper uses m, I think you need to modify the error in Github readme which is wrong in system of unit.

cattaneod, 2022-11-21:
> OK I know, the results you put in Github uses cm, and the results in your paper uses m, I think you need to modify the error in Github readme which is wrong in system of unit.

Hi @Zhaohx424,
you are right, the results in the readme use a wrong unit. thank you for pointing that out. I will correct it now.

closed 2022-12-02 by cattaneod