cattaneod/CMRNet #8: Problem with LocalMaps Generation

RaviBeagle, 2022-05-19:
Hello,

I am following the README.md and trying to generate the local maps with the given command:

**python preprocess/kitti_maps.py --sequence 00 --kitti_folder ./KITTI_ODOMETRY/**
Sequnce:  00
Traceback (most recent call last):
  File "preprocess/kitti_maps.py", line 54, in <module>
    kitti = pykitti.odometry(args.kitti_folder, sequence)
  File "/home/sxv1kor/anaconda3/envs/cmrnet/lib/python3.6/site-packages/pykitti/odometry.py", line 33, in __init__
    self._load_calib()
  File "/home/sxv1kor/anaconda3/envs/cmrnet/lib/python3.6/site-packages/pykitti/odometry.py", line 169, in _load_calib
    data['T_cam0_velo'] = np.reshape(filedata['Tr'], (3, 4))
KeyError: 'Tr'

All requirements of python environment are satisfied and I can see this line:

Requirement already satisfied: pykitti in /home/sxv1kor/anaconda3/envs/cmrnet/lib/python3.6/site-packages (from -r requirements.txt (line 11)) (0.3.1)

Any hints to solve this issue ? would be helpful.

Thanks

cattaneod, 2022-05-19:
You probably miss the KITTI calibration data: https://s3.eu-central-1.amazonaws.com/avg-kitti/data_odometry_calib.zip

RaviBeagle, 2022-05-19:
As far as I can see the KITTI data set sequence folders each have a calib.txt and times.txt already. Should I overwrite them with the data from the ones in https://s3.eu-central-1.amazonaws.com/avg-kitti/data_odometry_calib.zip ?

cattaneod, 2022-05-19:
Yes, the ones that are already there needs to be overwritten with the ones in data_odometry_calib.zip

RaviBeagle, 2022-05-19:
Thank you. There seems one minor issue in the scripts. The kitty_maps.py script requires the sequences to be in the folder './KITTI_ODOMETRY/sequences/'

The training and evaluation script requires the sequences to be in ./KITTI_ODOMETRY/ folder itself.

RaviBeagle, 2022-05-20:
> Yes, the ones that are already there needs to be overwritten with the ones in data_odometry_calib.zip

Thank you very much. Its working now. 
May I know if the current repository is of CMRNet or it already includes the CMRNet++ ? I did not see any references to the RANSAC and PnP algorithms..

RaviBeagle, 2022-05-20:
Also I would like to know if there is any hard dependency on a version of PyTorch. 
Because I am using a newer NVIDIA GPU and still getting a CUDA error:

Traceback (most recent call last):
File "preprocess/kitti_maps.py", line 108, in <module>
local_map = voxelized.clone()
RuntimeError: CUDA error: no kernel image is available for execution on the device

nvidia-smi
Fri May 20 10:55:13 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 510.73.05    Driver Version: 510.73.05    CUDA Version: 11.6     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA RTX A300...  Off  | 00000000:01:00.0  On |                  N/A |
| N/A   42C    P8    14W /  N/A |    588MiB /  6144MiB |      1%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

cattaneod, 2022-05-20:
> Thank you very much. Its working now. May I know if the current repository is of CMRNet or it already includes the CMRNet++ ? I did not see any references to the RANSAC and PnP algorithms..

This repository only includes CMRNet

> Also I would like to know if there is any hard dependency on a version of PyTorch. Because I am using a newer NVIDIA GPU and still getting a CUDA error:
> 
> Traceback (most recent call last): File "preprocess/kitti_maps.py", line 108, in local_map = voxelized.clone() RuntimeError: CUDA error: no kernel image is available for execution on the device
> 
> I see some discussion on the issue : #7 But I am not sure of the changes which needs to be done..
> 
> nvidia-smi Fri May 20 10:55:13 2022 +-----------------------------------------------------------------------------+ | NVIDIA-SMI 510.73.05 Driver Version: 510.73.05 CUDA Version: 11.6 | |-------------------------------+----------------------+----------------------+ | GPU Name Persistence-M| Bus-Id Disp.A | Volatile Uncorr. ECC | | Fan Temp Perf Pwr:Usage/Cap| Memory-Usage | GPU-Util Compute M. | | | | MIG M. | |===============================+======================+======================| | 0 NVIDIA RTX A300... Off | 00000000:01:00.0 On | N/A | | N/A 42C P8 14W / N/A | 588MiB / 6144MiB | 1% Default | | | | N/A | +-------------------------------+----------------------+----------------------+

Which pytorch version are you using? according to https://github.com/pytorch/pytorch/issues/49161, nvidia 3090 is only supported starting from pytorch 1.7.1.

RaviBeagle, 2022-05-20:
> 

I am using  pip show torch
Name: torch
Version: 1.0.1.post2

I have updated to the latest repository and I ran pip install -r requirements.txt again.
And I also just noticed that https://github.com/cattaneod/CMRNet documentation there is an additional step that is not mentioned in the README.md file:
viz. 
_"And finally, install the correlation_cuda and the visibility package:

cd models/CMRNet/correlation_package/
python setup.py install
cd ../../..
python setyp.py install"_

So I have done this step. And I also enabled :     '-gencode', 'arch=compute_80,code=sm_80',

Now I see a different error actually:

_THCudaCheck FAIL file=/pytorch/aten/src/THC/THCGeneral.cpp line=51 error=30 : unknown error
Traceback (most recent call last):
  File "preprocess/kitti_maps.py", line 94, in <module>
    voxelized = voxelized.to(args.device)
  File "/home/sxv1kor/anaconda3/envs/cmrnet/lib/python3.6/site-packages/torch/cuda/__init__.py", line 162, in _lazy_init
    torch._C._cuda_init()
RuntimeError: cuda runtime error (30) : unknown error at /pytorch/aten/src/THC/THCGeneral.cpp:51_

cattaneod, 2022-05-20:
You need to update pytorch to a newer version to run it with nvidia 3090.

I would suggest you to create a new conda environment and follow the new installation instructions from scratch. I'm sorry for your inconvenience.

RaviBeagle, 2022-05-20:
> You need to update pytorch to a newer version to run it with nvidia 3090.
> 
> I would suggest you to create a new conda environment and follow the new installation instructions from scratch. I'm sorry for your inconvenience.
> 
> PS: I would also suggest to clone the repository again, as you might have some old built files, and python would use them instead of build them again. Or, alternatively, remove any `build`, `dist` folder and any `*.so` file from the root of the repository and from `models/CMRNet/correlation_package`

No problem! I am doing just that now and I have to install torch, torchvision and cupy.
Is it fine if I install cupy-cuda102 ?

cattaneod, 2022-05-20:
> No problem! I am doing just that now and I have to install torch, torchvision and cupy. Is it fine if I install cupy-cuda102 ? I actually have RTX A3000 which is similar to RTX 3060..

No, you need to use the same cuda version for everything (pytorch, cupy, and so on), and it should be the version you have installed in your system. As the AMPERE architecture require CUDA 11.x, cupy-cuda102 won't work.

RaviBeagle, 2022-05-20:
> 

So, all right the commands should do the trick

nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Sun_Jul_28_19:07:16_PDT_2019
Cuda compilation tools, release **10.1**, V10.1.243

pip install cupy_cuda**101**=9.6.1
pip install torch==1.5.0+**cu101** torchvision==0.6.0+**cu101** -f

cattaneod, 2022-05-20:
I don't think you can use cuda 10.1 with your nvidia A3000, you can try, but if it doesn't work, you'll need to update cuda to (at least) 11

RaviBeagle, 2022-05-23:
> I don't think you can use cuda 10.1 with your nvidia A3000, you can try, but if it doesn't work, you'll need to update cuda to (at least) 11
> 
> https://docs.nvidia.com/cuda/ampere-compatibility-guide/index.html
> 
> > CUDA applications built using CUDA Toolkit versions 2.1 through 10.2 are compatible with NVIDIA Ampere architecture based GPUs as long as they are built to include PTX versions of their kernels.
> 
> I can't guarantee that it will works with cuda 10.1

Now I have updated the CUDA toolkit:
Device 0: "NVIDIA RTX A3000 Laptop GPU"
  CUDA Driver Version / Runtime Version          11.6 / 11.6

I am using Ubuntu 20.04 and I had to use a conda 3.7 environment because the CuPy requirement(https://docs.cupy.dev/en/stable/install.html).

Now the versions installed are:
# Name                    Version                   Build  Channel
torch                     1.12.0.dev20220520+cu116          pypi_0    pypi
torchvision               0.13.0.dev20220520+cu116          pypi_0    pypi
cupy-cuda116              10.4.0                   pypi_0    pypi

And these are working fine now:
cd models/CMRNet/correlation_package/
python setup.py install
cd ../../..
python setyp.py install

After this  the command 
'```python preprocess/kitti_maps.py --sequence 00 --kitti_folder ./KITTI_ODOMETRY/

gives error :

voxelized = torch.tensor(downpcd.points, dtype=torch.float)
0%| | 0/4541 [00:00<?, ?it/s]
Traceback (most recent call last):
File "preprocess/kitti_maps.py", line 108, in <module>
local_map = voxelized.clone()
**RuntimeError: CUDA error: __global__ function call is not configure**d
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.

RaviBeagle, 2022-05-25:
There is also some warning here, maybe a hint:

preprocess/kitti_maps.py:94: UserWarning: Creating a tensor from a list of numpy.ndarrays is extremely slow. Please consider converting the list to a single numpy.ndarray with numpy.array() before converting to a tensor. (Triggered internally at  ../torch/csrc/utils/tensor_new.cpp:201.)
  voxelized = torch.tensor(downpcd.points, dtype=torch.float)

RaviBeagle, 2022-05-25:
Hello @cattaneod 

I solved this problem by switching to CPU computation instead of GPU.

cattaneod, 2022-05-25:
I don't think it will work with cpu, some functions are implemented only in GPU.

To me it seems that your ptoblem is still cuda. Check the version with nvcc, not with `nvidia-smi`, as the latter will show the supported cuda version, not the installed version.

RaviBeagle, 2022-06-02:
> This repository only includes CMRNet

Is there a plan to also put CMRNet++ on github ?

RaviBeagle, 2022-06-15:
> 

I think there might be possibility of using CUDA 10.2 on Ampere Architecture. Will explore it:
https://docs.nvidia.com/cuda/ampere-compatibility-guide/index.html#verifying-ampere-compatibility-using-cuda-10-2

closed 2022-06-15 by RaviBeagle