cattaneod/CMRNet #10: Training fails after epoch 0

RaviBeagle, 2022-05-25:
The training on my NVIDIA A3000 GPU fails after Epoch 0.  I used batch size 8 since I was getting out of GPU memory error. System: Ubuntu 20.04.
Versions: 
torch                     1.12.0.dev20220520+cu116          pypi_0    pypi
torchvision               0.13.0.dev20220520+cu116          pypi_0    pypi
cupy-cuda116              10.4.0                   pypi_0    pypi
"python main_visibility_CALIB.py with batch_size=8 data_folder=./KITTI_ODOMETRY/ epochs=300 max_r=10 max_t=2 BASE_LEARNING_RATE=0.0001 savemodel=./checkpoints/ test_sequence=0" 
Is this a PyTorch error ?
Iter 1400/1429 training loss = 1.239, time = 0.0311, time for 50 iter: 12.7947
------------------------------------
epoch 0 total training loss = 1.323
Total epoch time = 358.35
------------------------------------
Init worker 0 with seed 889683110
Init worker 1 with seed 889683111
Init worker 2 with seed 889683112
ERROR - CMRNet - Failed after 0:06:03!
Traceback (most recent calls WITHOUT Sacred internals):
  File "main_visibility_CALIB.py", line 425, in main
    depth_img = visibility.depth_image(uv, depth, depth_img, uv.shape[0], real_shape[1], real_shape[0])
RuntimeError: input_uv.is_contiguous() INTERNAL ASSERT FAILED at "src/visibility.cpp":30, please report a bug to PyTorch. input_uv must be contiguous
(cmrnet2)

closed 2022-05-25 by cattaneod

fmaaf, 2023-09-28:
I met the same problem, do you have any idea about this? @cattaneod . Thanks.

fmaaf, 2023-09-28:
I have solved the problem: just replace _uv_ with _uv.contiguous()_.