cattaneod/CMRNet #9: Training fails after Epoch 0

RaviBeagle, 2022-05-25:
Hello @cattaneod ,

The training on my NVIDIA A3000 GPU fails after Epoch 0:

I used batch size 8 since I was getting out of GPU memory error. 

System: Ubuntu 20.04.

torch                     1.12.0.dev20220520+cu116          pypi_0    pypi
torchvision               0.13.0.dev20220520+cu116          pypi_0    pypi
cupy-cuda116              10.4.0                   pypi_0    pypi

python main_visibility_CALIB.py with batch_size=**8** data_folder=./KITTI_ODOMETRY/ epochs=300 max_r=10 max_t=2 BASE_LEARNING_RATE=0.0001 savemodel=./checkpoints/ test_sequence=0

Is this a PyTorch error ?

Iter 1400/1429 training loss = 1.239, time = 0.0311, time for 50 iter: 12.7947
------------------------------------
epoch 0 total training loss = 1.323
Total epoch time = 358.35
------------------------------------
Init worker 0 with seed 889683110
Init worker 1 with seed 889683111
Init worker 2 with seed 889683112
ERROR - CMRNet - Failed after 0:06:03!
Traceback (most recent calls WITHOUT Sacred internals):
  File "main_visibility_CALIB.py", line 425, in main
    depth_img = visibility.depth_image(uv, depth, depth_img, uv.shape[0], real_shape[1], real_shape[0])
RuntimeError: input_uv.is_contiguous() INTERNAL ASSERT FAILED at "src/visibility.cpp":30, p**lease report a bug to PyTorch. input_uv must be contiguous**
(cmrnet2)

cattaneod, 2022-05-25:
Replace `uv` with `uv.contiguous()`

RaviBeagle, 2022-05-25:
Yes. Now that error is no there anymore. But I get this strange error:

ERROR - CMRNet - Failed after 0:05:05!
Traceback (most recent calls WITHOUT Sacred internals):
  File "main_visibility_CALIB.py", line 465, in main
    sample['rot_error'], loss_fn, dataset_val.model, sample['point_cloud'])
  File "main_visibility_CALIB.py", line 126, in test
    total_trasl_error += torch.norm(target_transl[j] - transl_err[j]) * 100.
RuntimeError: Expected all tensors to be on the same device, but found at least two devices, cuda:0 and cpu!

I have not changed anything in the main_visibility_CALIB.py except the uv.contiguous() change..

RaviBeagle, 2022-05-25:
I cant be sure which of the Tensors are on CPU,
I printed out this and both are on CUDA.

target_transl[j] is on cuda:  True  and  transl_err[i] is on cuda:  True

cattaneod, 2022-05-25:
Change line 120 from `total_trasl_error = torch.tensor(0.0)` to `total_trasl_error = torch.tensor(0.0, device=target_transl.device)`

Sorry for the issues, I'll update the code to solve these issues

RaviBeagle, 2022-05-26:
Thanks. I am am able to run the training. 
No problem with the issues!  since I got an opportunity to read and try to understand  the cod. That will be useful.

closed 2022-06-02 by RaviBeagle