cattaneod/CMRNet #11: Problem with evaluation script

RaviBeagle, 2022-06-02:
Hello @cattaneod ,

I am getting an error from the sacred python module as follows:

<pre>python evaluate_iterative_single_CALIB.py with test_sequence=0 maps_folder=local_maps data_folder=./KITTI_ODOMETRY/sequences/ weight=&quot;[&apos;./checkpoints/iter1.tar&apos;,&apos;./checkpoints/iter2.tar&apos;,&apos;./checkpoints/iter3.tar&apos;]&quot;
Traceback (most recent call last):
  File &quot;evaluate_iterative_single_CALIB.py&quot;, line 35, in &lt;module&gt;
    ex = Experiment(&quot;CMRNet-evaluate-iterative&quot;)
  File &quot;/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/site-packages/sacred/experiment.py&quot;, line 80, in __init__
    _caller_globals=caller_globals)
  File &quot;/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/site-packages/sacred/ingredient.py&quot;, line 55, in __init__
    gather_sources_and_dependencies(_caller_globals, self.base_dir)
  File &quot;/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/site-packages/sacred/dependencies.py&quot;, line 487, in gather_sources_and_dependencies
    sources = gather_sources(globs, base_dir)
  File &quot;/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/site-packages/sacred/dependencies.py&quot;, line 438, in get_sources_from_imported_modules
    return get_sources_from_modules(iterate_imported_modules(globs), base_path)
  File &quot;/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/site-packages/sacred/dependencies.py&quot;, line 407, in get_sources_from_modules
    filename = os.path.abspath(mod.__file__)
  File &quot;/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/posixpath.py&quot;, line 378, in abspath
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType</pre>

Any idea how to to solve this issue ? Is this related to Python 3.7 ?

Thanks

RaviBeagle, 2022-06-03:
Hello @cattaneod,

I solved this issue by adding the following lines to evaluate script

<pre>SETTINGS.DISCOVER_DEPENDENCIES = &quot;none&quot;
SETTINGS.DISCOVER_SOURCES = &quot;none&quot;
</pre>

But now I get the following error:

<pre>python evaluate_iterative_single_CALIB.py with test_sequence=00 maps_folder=local_maps data_folder=./KITTI_ODOMETRY/sequences weight=&quot;[&apos;./checkpoints/iter1.tar&apos;,&apos;./checkpoints/iter2.tar&apos;,&apos;./checkpoints/iter3.tar&apos;]&quot;
WARNING - CMRNet-evaluate-iterative - No observers have been added to this run
INFO - CMRNet-evaluate-iterative - Running command &apos;main&apos;
INFO - CMRNet-evaluate-iterative - Started
TEST SET: Using this file: ./KITTI_ODOMETRY/sequences/test_RT_seq00_10.00_2.00.csv
ERROR - CMRNet-evaluate-iterative - Failed after 0:00:01!
Traceback (most recent calls WITHOUT Sacred internals):
  File &quot;evaluate_iterative_single_CALIB.py&quot;, line 107, in main
    test_sequence=_config[&apos;test_sequence&apos;])
  File &quot;/home/sxv1kor/Temp/CMRNet/DatasetVisibilityKitti.py&quot;, line 102, in __init__
    assert len(self.test_RT) == len(self.all_files), &quot;Something wrong with test RTs&quot;
AssertionError: Something wrong with test RTs</pre>

cattaneod, 2022-06-06:
Hi @RaviBeagle ,

the random H_init used for testing are generated the first time the evaluation is called, maybe something wrong happened when generating these transformations.

Try to remove the file `./KITTI_ODOMETRY/sequences/test_RT_seq00_10.00_2.00.csv` and see if it works.

RaviBeagle, 2022-06-09:
> Try to remove the file `./KITTI_ODOMETRY/sequences/test_RT_seq00_10.00_2.00.csv` and see if it works.

It does run now. But stops later:

<pre>ython evaluate_iterative_single_CALIB.py with test_sequence=00 maps_folder=local_maps data_folder=./KITTI_ODOMETRY/sequences/ weight=&quot;[&apos;./checkpoints/iter1.tar&apos;,&apos;./checkpoints/iter2.tar&apos;,&apos;./checkpoints/iter3.tar&apos;]&quot;
WARNING - CMRNet-evaluate-iterative - No observers have been added to this run
INFO - CMRNet-evaluate-iterative - Running command &apos;main&apos;
INFO - CMRNet-evaluate-iterative - Started
TEST SET - Not found: ./KITTI_ODOMETRY/sequences/test_RT_seq00_10.00_2.00.csv
Generating a new one
0
0it [00:00, ?it/s]Init worker 0 with seed 273103621
Init worker 1 with seed 273103622
Init worker 4 with seed 273103625
Init worker 3 with seed 273103624
Init worker 2 with seed 273103623
Init worker 5 with seed 273103626

Iterative refinement: 
Iteration 0: 	Mean Translation Error: nan cm      Mean Rotation Error: nan °
Iteration 0: 	Median Translation Error: nan cm      Median Rotation Error: nan °

Iteration 1: 	Mean Translation Error: nan cm      Mean Rotation Error: nan °
Iteration 1: 	Median Translation Error: nan cm      Median Rotation Error: nan °

Iteration 2: 	Mean Translation Error: nan cm      Mean Rotation Error: nan °
Iteration 2: 	Median Translation Error: nan cm      Median Rotation Error: nan °

Iteration 3: 	Mean Translation Error: nan cm      Mean Rotation Error: nan °
Iteration 3: 	Median Translation Error: nan cm      Median Rotation Error: nan °

-------------------------------------------------------
Timings:
ERROR - CMRNet-evaluate-iterative - Failed after 0:00:04!
Traceback (most recent calls WITHOUT Sacred internals):
  File &quot;evaluate_iterative_single_CALIB.py&quot;, line 431, in main
    errors_t2[i] = torch.stack(errors_t2[i])
RuntimeError: stack expects a non-empty TensorList
</pre>

The evaluation script does not need the training to be run first in anycase right ?

RaviBeagle, 2022-06-09:
> 

Working now. But I get the  old CUDA problem of not able to clone() on  GPU.

<pre>Init worker 4 with seed 399312879

ERROR - CMRNet-evaluate-iterative - Failed after 0:00:04!
Traceback (most recent calls WITHOUT Sacred internals):
  File &quot;evaluate_iterative_single_CALIB.py&quot;, line 308, in main
    rotated_point_cloud = rotate_forward(point_cloud, RT1)
  File &quot;/home/sxv1kor/Temp/CMRNet/utils.py&quot;, line 79, in rotate_forward
    return rotate_points_torch(PC, R, T, inverse=True)
  File &quot;/home/sxv1kor/Temp/CMRNet/utils.py&quot;, line 46, in rotate_points_torch
    RT = R.clone()
RuntimeError: CUDA error: __global__ function call is not configured
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.</pre>

cattaneod, 2022-06-09:
I never encounter that problem, so I don't know how to help you. Have you installed cnDNN? I'm not even sure that it's required, but maybe try to install it if you didn't already.

RaviBeagle, 2022-06-09:
> 

Yes. Its a very strange error. I will look to PyTorch forums for help. 
cuDNN is installed in my system

RaviBeagle, 2022-06-10:
> I never encountered that problem, so I don't know how to help you. Have you installed cnDNN? I'm not even sure if that's required, but maybe try to install it if you didn't already.

Is it possible to run the evaluation on CPU ?

RaviBeagle, 2022-06-15:
> I never encountered that problem, so I don't know how to help you. Have you installed cnDNN? I'm not even sure if that's required, but maybe try to install it if you didn't already.

And now its working. I changed to Python 3.6 and CUDA 10.2 and installed the pytorch with command:
conda install pytorch torchvision torchaudio cudatoolkit=10.2 -c pytorch

Everything is working fine now..

cattaneod, 2022-06-15:
> And now its working. I changed to Python 3.6 and CUDA 10.2 and installed the pytorch with command: conda install pytorch torchvision torchaudio cudatoolkit=10.2 -c pytorch
> 
> And also matching cudnn as specified in the pytorch
> 
> Everything is working fine now..

I'm glad that now it's working fine!