cattaneod/CMRNet #12: About model performance

WJ1214, 2022-06-06:
Hi, I noticed that the results you provide in this repo are:
[image]
While the results you provide in your paper are:
[image]
It seems like there are two different unit of translation error in repo and paper, which one is correct?

WJ1214, 2022-06-06:
Besides, I'm also confusing about the code in training scipt here:
[image]
It seems like used the X ocordinate of point cloud to filter the point which is far more than max_depth, why?

cattaneod, 2022-06-06:
Hi @WJ1214 ,

regarding your first question, you're comparing the results reported in this repo (CMRNet), with the ones reported in the CMRNet++ paper, as they are two different approaches, the results are different.

Regarding your second question, as we project the map (and not a single point cloud) into the virtual image plane, the points could be potentially infinitely far away, that's why we need to set a maximum distance.

closed 2022-10-17 by cattaneod