cattaneod/CMRNet #13: Using a different data set for training and evaluation

RaviBeagle, 2022-06-17:
Hello @cattaneod ,

How would you recommend as the steps required to use a different data set for training and evaluating the CMRNet ?
I understand that CMRNet is scene agnostic and I could in theory evaluate the pre-trained model against a new dataset.

Thanks

cattaneod, 2022-06-17:
Hi @RaviBeagle ,

although CMRNet is scene agnostic, it's not camera agnostic, which means that it can only be deployed using the same camera sensor as used during training. To overcome this issue, we proposed CMRNet++ which is scene and sensor agnostic, but the code is not publicly available at the moment.

In order to train CMRNet on a different dataset, the first and most important step is to check the quality of the ground truth poses. If the GT poses are not accurate, you probably won't be able to train CMRNet.
Then, you should preprocess the dataset, by generating the map of every sequence, and generating a local map for each camera image, as done in https://github.com/cattaneod/CMRNet/blob/master/preprocess/kitti_maps.py. As mentioned in the README, the local maps should have this reference frame: X-forward, Y-right, Z-down.

After the dataset is preprocessed, you should adapt the data loader (https://github.com/cattaneod/CMRNet/blob/master/DatasetVisibilityKitti.py), make sure to change the camera calibration to your own calibration parameters (the images should also be undistorted beforehand).

Finally, you should change the image_size parameter to a suitable size for you images (take into account that both width and height should be multiple of 64, due to the architecture of the network).

RaviBeagle, 2022-07-21:
Thanks a lot for the hints. 
We are now preparing our vehicle setup to capture the dataset. The sensor setup is as follows:
1) Velodyne HDL-16 as our LiDAR for Localization. Driving area has already has the point cloud maps generated with same sensor. 
2) MYNTEYE S1030 Stereo camera. This camera has provides grayscale images.

The plan is to run the ROS HDL localization package to generate the GT poses and capture the sensor data with approximate time synchronization.  Do you see any limitations of our setup ?

cattaneod, 2022-07-28:
I'm not familiar with the ROS HDL package, but if the generated GT poses are accurate enough, I don't see any big limitations.
CMRNet is intended for monocular localization, since you have a stereo setup, I can guess that you can further improve the localization performance by including the second camera.

RaviBeagle, 2022-08-16:
Hello @cattaneod ,
Due to delays in setting up our vehicle and sensors, we plan in the meantime to test out a synthetic dataset from CARLA (https://github.com/jedeschaud/kitti_carla_simulator) to train. With some modifications to the scripts we hope to have a dataset similar to KITTI.  The CMRNet would not need any modifications in that case, if I am right ?

Thanks

RaviBeagle, 2022-09-09:
Hello @cattaneod ,

Have managed to generate KITTI type of dataset from the CARLA simulator. Here one question on a comment you put in the documentation:

"The Data Loader requires a local point cloud for each camera frame, the point cloud must be expressed with respect to the camera_2 reference frame, BUT (very important) with a different axes representation: X-forward, Y-right, Z-down."

Is this done by the preprocess/kitti_maps.py or is it something I have to do at the time of generation ?

cattaneod, 2022-09-09:
Yes, it is done in [kitti_maps.py#L119](https://github.com/cattaneod/CMRNet/blob/master/preprocess/kitti_maps.py#L119)

RaviBeagle, 2022-09-11:
> Finally, you should change the image_size parameter to a suitable size for you images (take into account that both width and height should be multiple of 64, due to the architecture of the network).

I have set the RGB image size in CARLA as 1344x512 which is multiple of 64 

Yet I get some error during training:

```
File "/home/sxv1kor/Temp/CMRNet/DatasetVisibilityKitti.py", line 180, in __getitem__
    img = self.custom_transform(img, img_rotation, h_mirror)
  File "/home/sxv1kor/Temp/CMRNet/DatasetVisibilityKitti.py", line 135, in custom_transform
    rgb = normalization(rgb)
  File "/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/site-packages/torchvision/transforms/transforms.py", line 269, in forward
    return F.normalize(tensor, self.mean, self.std, self.inplace)
  File "/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/site-packages/torchvision/transforms/functional.py", line 360, in normalize
    return F_t.normalize(tensor, mean=mean, std=std, inplace=inplace)
  File "/home/sxv1kor/anaconda3/envs/cmrnet2/lib/python3.7/site-packages/torchvision/transforms/functional_tensor.py", line 959, in normalize
    tensor.sub_(mean).div_(std)
RuntimeError: The size of tensor a (4) must match the size of tensor b (3) at non-singleton dimension 0

cattaneod, 2022-09-11:
As the error says, your images have 4 channels instead of 3 [R,G,B].

Sorry, I can't really help you with different datasets.

closed 2022-09-11 by cattaneod

RaviBeagle, 2022-09-15:
Yes, indeed that was the problem. Thanks.
I have saved the images as RGB and now the training is happening.

RaviBeagle, 2022-10-14:
> Yes, it is done in [kitti_maps.py#L119](https://github.com/cattaneod/CMRNet/blob/master/preprocess/kitti_maps.py#L119)

The line 119:
    local_map = local_map[[2, 0, 1, 3], :]

LIDAR: x: forward, y: left and z: up
Required: x: forward, y: right, z: down

Should we not just negate y and z to get to that representation ?

cattaneod, 2023-05-25:
> Hi @cattaneod , Thanks for your amazing job. I wonder how you fetch the surrounding point cloud that forms the local submap? Specifically, I mean [here](https://github.com/cattaneod/CMRNet/blob/master/preprocess/kitti_maps.py#L111). The code confuses me quite a lot.

What exactly is not clear?
To generate a local submap around the camera pose, I first transform the global map into a local map, such that the camera pose is the origin of the point cloud

 ```python
local_map = torch.mm(pose, local_map).t()`
```

Then, I crop the point cloud around the origin (which now is the camera pose). Specifically, i take 25 meters on the left and right, 100 meters in the front, and -10 in the back, to account for the random initial position H_init that could be behind the real pose.

 ```python
indexes = local_map[:, 1] > -25.  # Crop 25 meters to the right
indexes = indexes & (local_map[:, 1] < 25.)  # Crop 25 meters to the left
indexes = indexes & (local_map[:, 0] > -10.)  # Crop 10 meters to the back
indexes = indexes & (local_map[:, 0] < 100.)  # Crop 100 meters to the front
```

whu-lyh, 2023-05-26:
Ohhhhh, I figure it out. Thanks very much!

jiheddachraoui, 2025-03-05:
Good morning,

I tested your approach using a dataset I collected personally last year, which I formatted in the KITTI format. However, I did not obtain good results, with an error of around 200 cm. What do you think could be the issue? Could it be due to inaccuracies in the ground truth, calibration data, or data synchronization?

Even if these factors are not perfect, they would affect the error magnitude, but the algorithm should still visually produce a good correspondence between the image and LiDAR data. I also tried using the ground truth as input (with max_t=0 and max_r=0), yet there is still a slight misalignment between the image and LiDAR. The ground truth poses were extracted using Faster LIO, but shouldn’t the model correct this in the output to better align the points with the image? Independently of the error magnitude—which is expected to be large in this case—I am primarily interested in the visual alignment.

Personally, I suspect an issue with the calibration data, but I don’t believe it is significantly inaccurate enough to cause such an error.

By the way, I also tested your approach with KITTI data, and it worked as expected, as stated in the paper. Thank you for your work!

cattaneod, 2025-03-05:
Dear @jiheddachraoui ,

did you train CMRNet on your dataset, or did you evaluate the pre-trained model provided in the repository?

PS: very soon (ideally by the end of the month), I will release our latest method CMRNext (https://arxiv.org/abs/2402.00129), which achieves much better results, and can generalize much better.

jiheddachraoui, 2025-03-05:
Hello,

Yes, I have read your work on CMRNext, and I really appreciate the idea of it being independent of calibration data. I would love to test it on my own dataset and use it to validate my calibration.

To answer your question, no, I didn’t train the model—I was mainly interested in evaluating how well it generalizes. So, I only ran an evaluation using the pre-trained model you provided. Additionally, I’m not entirely sure about the accuracy of my ground truth for training.

Would you still recommend that I try training it anyway?

cattaneod, 2025-03-05:
Dear @jiheddachraoui ,

as mentioned in our paper, CMRNet is dependent on the intrinsic parameter of the camera used during training, and thus cannot generalize to different cameras. If you want to use it with your own sensor setup, you would need to train it from scratch.

However, I'd suggest to wait for the release of CMRNext, which should be able to perform well on unseen cameras out of the box.

jiheddachraoui, 2025-03-05:
Thank you again! I'll be waiting as I continue testing the state of the art.

jiheddachraoui, 2025-04-11:
Hello, I just wanted to share some information. The camera's intrinsic parameters are adjustable in the code, but one aspect I overlooked was the image size. The image size I worked with differed from the one used in KITTI, which I didn’t notice initially—only when I tested with very large images (from the Boreas dataset) did the issue become apparent. The image size used in the inference code is fixed, so when I worked with different sizes, the code only used the fixed number of pixels. This is why I noticed the LiDAR-to-image projection wasn’t accurate. The intrinsic matrix needs to be adjusted to account for the new image size, as cropping and resizing affect the intrinsic calibration. After I updated my intrinsic matrix, the projection worked correctly. However, the results are still far from the KITTI results, and I haven’t had time to investigate further—perhaps due to the snowy scenario I worked with.

I'm sharing this to help future testers avoid the same issue and save time.