cattaneod/CMRNet #15: Some hints needed on converting to online use

RaviBeagle, 2022-11-25:
Hello @cattaneod ,

The results I have obtained from training CMRNet with KITTI type of dataset from CARLA are quite encouraging.
Now I think I could really test CMRNet as a localization module in our test vehicle online. I am in process and I think understanding the evaluate_iterative_single_CALIB.py and simplifying it is the way to go. 

 What I plan to do is at each instance of the simulator running at 10 FPS:
1) Get the GT pose. 
2) Apply a random error to R and T to the GT pose and then
3) With just Iteration 1 of the CMRNet see how much of correction happens.

After this I could check with the same process on the vehicle. 

But there seems a lot going on in evaluate_iterative_single_CALIB.py with evaluation and visualization in that code. Some hints to the most essential part of the code would be helpful..

Thanks

RaviBeagle, 2022-11-28:
To my understanding the  evaluate_iterative_single_CALIB.py script use the testRT random translation and rotation error to do the evaluation. For the online usecase starting from a estimated pose, we would predict the transform to the actual pose. Then use that pose in next step successively. 

Initialize: set P0 <- Initial Estimated Pose

Loop at 100ms:
    1) Set P_Estimated =  P0
    2) Set P_real = Pose after CMRNet's predicted correction to the P0
    3) Set P_Estimated = P_real
    4) Goto Step 1

RaviBeagle, 2022-11-28:
Hello @cattaneod 

I am trying to check the projection of GT lidar to the camera with the following code:

```
    for i in tqdm(range(0,len(poses))):
        lidar_input = []
        rgb_input = []
        shape_pad = [0, 0, 0, 0]

        pose = poses[i]
        pose = pose.to(args.device)
        pose = pose.inverse()
        local_map = torch.mm(pose, voxelized).t()
        indexes = local_map[:, 1] > -25.
        indexes = indexes & (local_map[:, 1] < 25.)
        indexes = indexes & (local_map[:, 0] > -10.)
        indexes = indexes & (local_map[:, 0] < 100.)
        local_map = local_map[indexes]
        local_map = torch.mm(velo2cam2, local_map.t())
        local_map = local_map[[2, 0, 1, 3], :]

        cam_model = CameraModel()
        cam_model.focal_length = cam_params[:2]
        cam_model.principal_point = cam_params[2:]
        uv, depth, points, refl = cam_model.project_pytorch(local_map, real_shape, reflectance=None)
        uv = uv.t().int()
        depth_img = torch.zeros(real_shape[:2], device='cuda', dtype=torch.float)
        depth_img += 1000.
        depth_img = visibility.depth_image(uv.contiguous(), depth, depth_img, uv.shape[0], real_shape[1], real_shape[0])
        depth_img[depth_img == 1000.] = 0.

        projected_points = torch.zeros_like(depth_img, device='cuda')
        projected_points = visibility.visibility2(depth_img, cam_params, projected_points, depth_img.shape[1],
                                                      depth_img.shape[0], occlusion_threshold,occlusion_kernel)
        projected_points /= 100.
        projected_points = projected_points.unsqueeze(0)
        rgb_image  = image_to_tensor(kitti.get_cam2(i)).cuda()
        shape_pad[3] = (img_shape[0] - rgb_image.shape[1])
        shape_pad[1] = (img_shape[1] - rgb_image.shape[2])
        rgb_image = F.pad(rgb_image, shape_pad)
        projected_points = F.pad(projected_points, shape_pad)

        rgb_input.append(rgb_image)
        lidar_input.append(projected_points)

        lidar_input = torch.stack(lidar_input)
        rgb_input = torch.stack(rgb_input)

        out0 = overlay_imgs(rgb_image, lidar_input)
        cv2.imshow("INPUT", out0[:, :, [2, 1, 0]])
        cv2.waitKey(1)
```

But I see some sort of mirror imaging of the lidar with respect to camera. Why is this occuring.

cattaneod, 2022-12-02:
Your reasoning sounds correct to me.
However, I believe that you'll need to change how the map is loaded and processed. The provided code assumes that the point clouds are expressed in the camera reference frame (i.e., the ground truth pose is the identity matrix). However, in the online setting, you don't know the poses of the camera, and probably you'll need to load the full map point cloud, and use "global" poses instead of local poses.

RaviBeagle, 2022-12-02:
Hello @cattaneod ,

Yes. I do load the entire point cloud map and calculate the local maps in camera reference frame (with appropriate axis shifting as required) online. Also the tr_error and rot_error is calculated per pose randomly.  The code is completed and working well. However I still take the poses from file. I will send the code by mail. 

I did not understand the point you mentioned on "global" poses and "local" poses. Currently this problem I dont see as in CARLA simulation where I am testing the poses provided are always the global ones. In the vehicle we might have to convert GPS to local coordinates. 

By second attempt at using CMRNet to directly estimate the odometry i.e. using the loop I mentioned:

Initialize: set P0 <- Initial Estimated Pose of Lidar (based on converting GPS to Local Co-ordinates)
1) Set P_Estimated = P0

Loop at 100ms:
2) Set P_real = Pose after CMRNet's predicted correction to the P0
3) Set P_Estimated = P_real
4) Goto Step 2

This is not working yet.

cattaneod, 2022-12-06:
You just need to apply the predicted transformation to the initial global pose.
RT_global = RT_init @ RT_predicted

RaviBeagle, 2022-12-06:
> RT_global = RT_init @ RT_predicted

Do you mean to that if the RT_predicted is applied successively to the global pose (the camera global poses or lidar global poses ?) its should be enough 

# Apply at each step
RT_global_predicted = RT_global_estimated @ RT_predicted  

But when I read the code in evaluate_iterative_single_CALIB.py, I see that the RT_predicted is a transform that applies to the local point cloud map in different axes representation (i.e, x-right, y-forward,z-down ?) Can we use that directly on global poses in a different frame and axes ?

cattaneod, 2022-12-06:
You might be right, the RT_predicted might be in the different axis representation. In that case you should additionally apply the transformation to convert the axis of the local map (X-forward, Y-right, Z-down), to the axis used in your global map

RaviBeagle, 2022-12-06:
> 
Strangely, if I do that its seems to be actually making the errors (euclidean distance between GT pose and predicted pose) worse.

RaviBeagle, 2022-12-06:
Hello @cattaneod 

I think I figured it. Please correct me if wrong:

First I convert the global estimated lidar pose into local frame after velo2cam and cam2projection conversion
```
        map_cam2_frame = torch.mm(velo2cam2,pose_estimated_lidar)
        map_cam2_frame = torch.mm(cam2projection,map_cam2_frame)
        map_cam2_frame = map_cam2_frame.inverse()
```

After that I set
            pose_predicted_lidar = map_cam2_frame

and within the prediction loop I apply:

```              
 pose_predicted_lidar = torch.mm(RT_predicted,pose_predicted_lidar)
```

After that I convert the pose in local frame back to global frame:

```
        pose_predicted_lidar = pose_predicted_lidar.inverse()
        pose_predicted_lidar = torch.mm(cam2projection.inverse(),pose_predicted_lidar)
        pose_predicted_lidar = torch.mm(velo2cam2.inverse(), pose_predicted_lidar)
```

I am also calculating the euclidean stance between predicted pose and GT pose: 

```
        x_diff = abs((poses[i][0, 3]-pose_predicted_lidar[0, 3]).item())
        y_diff = abs((poses[i][1, 3]-pose_predicted_lidar[1, 3]).item())
        z_diff = abs((poses[i][2, 3]-pose_predicted_lidar[2, 3]).item())

        euclidean_dist = math.sqrt(x_diff**2 + y_diff**2 + z_diff**2)
        print(euclidean_dist)
```

I am getting these values

3.542215426981913
3.080128885975539
3.067205507491944
1.1507081310578107
1.1469494296990475
3.5180695309269807
1.3522896182893547
2.8925430647598285
1.7787128782295187
1.744652965101788
3.047048332904441
1.704646785048259

rreddy78, 2022-12-12:
> You might be right, the RT_predicted might be in the different axis representation. In that case you should additionally apply the transformation to convert the axis of the local map (X-forward, Y-right, Z-down), to the axis used in your global map

Hello @cattaneod 

Yes.! I used this and its working now. I am getting the following values as the euclidean distances. It looks much better than before

1.0016057015709217
0.7632609130110889
1.6000662965492867
0.6335314340250894
1.4316691714409067
0.4104338104571504
0.5412108653227982
0.8677367236067993
0.06451896243708906
0.689013571125916
0.18936646651593728
1.1538748902937705
1.6963246365141917
1.5277596833268183
0.7666823448536583
1.516898214231317
1.4226326021778204
1.7708160155482124
1.203460323053111

RaviBeagle, 2022-12-12:
> You might be right, the RT_predicted might be in the different axis representation. In that case you should additionally apply the transformation to convert the axis of the local map (X-forward, Y-right, Z-down), to the axis used in your global map

Hello @cattaneod ,

Yes! I did this and its working good now. I get these values are euclidean distances:

Hello @cattaneod

Yes.! I used this and its working now. I am getting the following values as the euclidean distances. It looks much better than before

1.0016057015709217
0.7632609130110889
1.6000662965492867
0.6335314340250894
1.4316691714409067
0.4104338104571504
0.5412108653227982
0.8677367236067993
0.06451896243708906
0.689013571125916
0.18936646651593728
1.1538748902937705
1.6963246365141917
1.5277596833268183
0.7666823448536583
1.516898214231317
1.4226326021778204
1.7708160155482124
1.203460323053111

And the values look much better!

RaviBeagle, 2022-12-15:
Hello @cattaneod 

While values look better I was expecting always to get values < 20cm atleast. But there are lots of jumps and inconsistencies. I am running iteratively with all three models of KITTI (provided by you) with sequence 00.

closed 2022-12-15 by RaviBeagle

cattaneod, 2022-12-15:
As written in the README and in our paper, CMRNet achieves 20cm MEDIAN translation error, and not maximum error, which means that for half of the frames the error will be higher than 20cm, and for the other half it'll be lower than 20cm.

CMRNet was designed to perform localization, I never tested it for odometry, I'm sorry to hear that it doesn't perform as good as expected. Maybe the evolved CMRNet++ version might improve, but the code is not publicly available at the moment.