cattaneod/CMRNet #18: How can I apply the error_tr and error_rot for pose ?

RaviBeagle, 2022-12-06:
Hello @cattaneod 

I see that a random error_tr and error_rot can be applied to the local point cloud map as follows: 

```
        max_angle = 10.0
        max_t = 2.0
        rotz = np.random.uniform(-max_angle, max_angle) * (3.141592 / 180.0)
        roty = np.random.uniform(-max_angle, max_angle) * (3.141592 / 180.0)
        rotx = np.random.uniform(-max_angle, max_angle) * (3.141592 / 180.0)
        transl_x = np.random.uniform(-max_t, max_t)
        transl_y = np.random.uniform(-max_t, max_t)
        transl_z = np.random.uniform(-max_t, min(max_t, 1.))

        R = mathutils.Euler((rotx, roty, rotz), 'XYZ')
        T = mathutils.Vector((transl_x, transl_y, transl_z))
        R_inv, T_inv = invert_pose(R, T)
        R = mathutils.Quaternion(R_inv)
        T = mathutils.Vector(T_inv)
        rotated_point_cloud = rotate_back(local_map.clone(), R, T)
```

I guess this error_tr and error_rot are with respect to local point cloud map reference frame. How can apply the same for global poses in lidar reference frame ? I guess it has something to do with changing 

```
        transl_z = np.random.uniform(-max_t, min(max_t, 1.))
```
But not sure. 

Thanks

cattaneod, 2022-12-06:
When using a "global" map, you first need to convert it to a local frame, by applying the inverse of the global lidar pose, and then apply the augmentation (if needed).

RaviBeagle, 2022-12-06:
Ok. Thank you. Actually I wanted have the global estimated pose directly. I think this code is working.

```
        rotz = np.random.uniform(-max_angle, max_angle) * (3.141592 / 180.0)
        roty = np.random.uniform(-max_angle, max_angle) * (3.141592 / 180.0)
        rotx = np.random.uniform(-max_angle, max_angle) * (3.141592 / 180.0)
        transl_x = np.random.uniform(-max_t, max_t)
        transl_y = np.random.uniform(-max_t, max_t)
        transl_z = 0.0 #np.random.uniform(-max_t, min(max_t, 1.))

        R = mathutils.Euler((rotx, roty, rotz), 'XYZ')
        T = mathutils.Vector((transl_x, transl_y, transl_z))
        R = R.to_matrix()
        R.resize_4x4()
        T = mathutils.Matrix.Translation(T)
        RT = T * R
        pose_error_transform =  torch.tensor(RT)
        pose_estimated_lidar = torch.mm(poses[i],pose_error_transform)
        print("::",poses[i],"\n",pose_error_transform,"\n",pose_estimated_lidar)
        pose_estimated_lidar = pose_estimated_lidar.to(device
```
After getting the estimated global lidar pose I use that pose to compute the local lidar map like you said

```
        local_map = torch.mm(pose_estimated_lidar.inverse(), voxelized).t()
        indexes = local_map[:, 1] > -25.
        indexes = indexes & (local_map[:, 1] < 25.)
        indexes = indexes & (local_map[:, 0] > -10.)
        indexes = indexes & (local_map[:, 0] < 100.)
        local_map = local_map[indexes]
        local_map = torch.mm(velo2cam2, local_map.t())
        local_map = local_map[[2, 0, 1, 3], :]
```

I think my mistake was in this line. I had reversed the matrix multiplication
```
  pose_estimated_lidar = torch.mm(poses[i],pose_error_transform)
```

closed 2022-12-06 by RaviBeagle