cattaneod/CMRNet #20: Ground truth test pose is required

jinyummiao, 2022-12-22:
Hi, thanks for your insteresting work! I have evaluated your provided pretrained weights, but the performances cannot be reproduced. I got the performance as:
 
[image]
 
, which is much lower than your performance. I guess it may be caused by different ground-truth pose during evaluation? So, could you provide your test_RT_seq00_*_*.csv? 
Thanks at advance.

cattaneod, 2022-12-22:
Dear @jinyummiao ,

the pretrained weights provided  in this repo achieves slightly better performance than the results reported in the paper due to small improvements made after the paper submission.
As mentioned in the README, the provided weights achieved a median translation and rotation errors of 0.20m and 0.97°.
Indeed, your results are slightly better than the one reported, probably due to the different random initial position applied to the test poses, as you mentioned.

However, I'm afraid that I don't have the test poses used to achieved the results anymore, I'm sorry.

jiheddachraoui, 2025-02-28:
Hello, I'm trying to run the test directly with the pretrained weights, and I got this error. Do you know what the problem is?

```
WARNING - CMRNet-evaluate-iterative - No observers have been added to this run
INFO - CMRNet-evaluate-iterative - Running command 'main'
INFO - CMRNet-evaluate-iterative - Started
TEST SET - Not found: ./data/KITTI_ODOMETRY/sequences/test_RT_seq00_10.00_2.00.csv
Generating a new one
0
0it [00:00, ?it/s]Init worker 1 with seed 249601207
Init worker 2 with seed 249601208
Init worker 4 with seed 249601210
Init worker 5 with seed 249601211
Init worker 0 with seed 249601206
Init worker 3 with seed 249601209

Iterative refinement: 
Iteration 0: 	Mean Translation Error: nan cm      Mean Rotation Error: nan °
Iteration 0: 	Median Translation Error: nan cm      Median Rotation Error: nan °

Iteration 1: 	Mean Translation Error: nan cm      Mean Rotation Error: nan °
Iteration 1: 	Median Translation Error: nan cm      Median Rotation Error: nan °

Iteration 2: 	Mean Translation Error: nan cm      Mean Rotation Error: nan °
Iteration 2: 	Median Translation Error: nan cm      Median Rotation Error: nan °

Iteration 3: 	Mean Translation Error: nan cm      Mean Rotation Error: nan °
Iteration 3: 	Median Translation Error: nan cm      Median Rotation Error: nan °

-------------------------------------------------------
Timings:
ERROR - CMRNet-evaluate-iterative - Failed after 0:00:07!
Traceback (most recent calls WITHOUT Sacred internals):
  File "evaluate_iterative_single_CALIB.py", line 429, in main
    errors_t2[i] = torch.stack(errors_t2[i])
```
RuntimeError: stack expects a non-empty TensorList

cattaneod, 2025-02-28:
Dear @jiheddachraoui ,

there is an issue with the data structure.
As you can see, in the line below "Generating a new one", there is a 0, which means that the data loader couldn't find the proper data to load. That number should be around 4541 for sequence 00 of the kitti dataset.

Make sure that your folder `./data/KITTI_ODOMETRY` follows the structure provided in the readme of the repository:

```
KITTI_ODOMETRY
├── 00
│   ├── image_2
│   │   ├── 000000.png
│   │   ├── 000001.png
│   │   ├── ...
│   │   └── 004540.png
│   ├── local_maps
│   │   ├── 000000.h5
│   │   ├── 000001.h5
│   │   ├── ...
│   │   └── 004540.h5
│   └── poses.csv
└── 03
    ├── image_2
    │   ├── 000000.png
    │   ├── 000001.png
    │   ├── ...
    │   └── 000800.png
    ├── local_maps
    │   ├── 000000.h5
    │   ├── 000001.h5
    │   ├── ...
    │   └── 000800.h5
    └── poses.csv
```

Additionally, before you run the test script again, please delete the file `./data/KITTI_ODOMETRY/sequences/test_RT_seq00_10.00_2.00.csv`

cattaneod, 2025-02-28:
Now that I look closely, I think that a "sequences" folder is missing in the folder structure:
```
KITTI_ODOMETRY
├── sequences
│   ├── 00
│   │   ├── image_2
│   │   │   ├── 000000.png
│   │   │   ├── 000001.png
│   │   │   ├── ...
│   │   │   └── 004540.png
│   │   ├── local_maps
│   │   │   ├── 000000.h5
│   │   │   ├── 000001.h5
│   │   │   ├── ...
│   │   │   └── 004540.h5
│   │   └── poses.csv
│   └── 03
    │   ├── image_2
    │   │   ├── 000000.png
    │   │   ├── 000001.png
    │   │   ├── ...
    │   │   └── 000800.png
    │   ├── local_maps
    │   │   ├── 000000.h5
    │   │   ├── 000001.h5
    │   │   ├── ...
    │   │   └── 000800.h5
    │   └── poses.csv
```

jiheddachraoui, 2025-02-28:
Hello, thank you for the quick response the probleme where local_maps that have a number in the name i wasnt carefull.
thanks for this intresting work.

jiheddachraoui, 2025-02-28:
I encountered another error with PyTorch tensors, but I'll take my time to resolve it. I'll reach out if I need further assistance. Thank you again!