cattaneod/CMRNet #21: Occlusion filtering (train vs eval show)

christian-hiebl, 2022-12-27:
I have a question regarding the occlusion filtering. In the training part of CMRNet, there is no maxpooling done to filter out points behind other points, since this is done through this cone creation and filtering implemented in the visibility package, from my understanding. My question now is, why is there a maxpooling applied when we want to visualize the rgb and lidar points overlaid, as in 
https://github.com/cattaneod/CMRNet/blob/3359a2c85cbc7e270d499dd15b892f6bf058aac3/utils.py#L275

which is used here
https://github.com/cattaneod/CMRNet/blob/3359a2c85cbc7e270d499dd15b892f6bf058aac3/evaluate_iterative_single_CALIB.py#L257

Thank you for your great repository and support!

christian-hiebl, 2022-12-28:
I think it is for clearer visualization of the depth map, will close the issue now.

closed 2022-12-28 by christian-hiebl

cattaneod, 2023-01-09:
Your intuition is correct, the max pooling is used for better visualization of the depth map overlay with the camera image.